paulvha/sps30 #2: sps30.h >> SER_STOP_MEASUREMENT

detamend, 2019-04-04:
Hello Paul - great library!!

I may have stumbled upon a error in sps30.h, line 291:

> bool probe();
>    bool reset() {return(Instruct(SER_RESET));}
>    bool start() {return(Instruct(SER_START_MEASUREMENT));}
>    bool stop() {return(Instruct(SER_START_MEASUREMENT));}
>    bool clean() {return(Instruct(SER_START_FAN_CLEANING));}
		
should 

bool stop() {return(Instruct(SER_**START**_MEASUREMENT));}

not be

bool stop() {return(Instruct(SER_**STOP**_MEASUREMENT));}

? Seems like a copy-error :)

paulvha, 2019-04-04:
Good catch.. you are right.. I will make an update
Thanks
Paul

paulvha, 2019-04-04:
change applied.

Thanks again

regards,
Paul

closed 2019-04-04 by paulvha