paulvha/sps30 #3: PM2.5 - PM10 identical after stabilization time

jhinderer, 2019-04-06:
Hello and thank you for this great library!
I am using it with an ESP32 over UART and have a weird problem. After starting the measurement mode, it only takes some measurements (between 1 and 10) and the values for PM2.5, PM4 and PM10 concentrations are identical. I think it's pretty unlikely that there aren't any larger particles for several days. After a reset the first measurements show different values but they are getting the same pretty fast again.
Have you ever seen a similar behaviour?  I am not sure if my sensor is faulty or if I am doing anything wrong.

```
-------------Mass -----------    ------------- Number --------------   -Average-
     Concentration [μg/m3]             Concentration [#/cm3]             [μm]
P1.0	P2.5	P4.0	P10	P0.5	P1.0	P2.5	P4.0	P10	PartSize

4.41	7.04	8.87	9.25	24.16	32.17	34.83	35.15	35.21	0.75
4.22	5.75	6.77	6.98	26.03	32.34	33.86	34.04	34.07	0.65
4.02	4.16	4.16	4.16	28.70	32.90	33.00	33.00	33.00	0.53
3.89	4.03	4.03	4.03	27.75	31.82	31.91	31.91	31.91	0.51
3.75	3.88	3.88	3.88	26.74	30.66	30.75	30.75	30.75	0.53
3.98	4.12	4.12	4.12	28.39	32.56	32.65	32.65	32.65	0.50
3.98	4.12	4.12	4.12	28.41	32.57	32.66	32.66	32.66	0.50
3.94	4.08	4.08	4.08	28.10	32.22	32.31	32.31	32.32	0.48
3.85	3.99	3.99	3.99	27.47	31.49	31.58	31.58	31.59	0.47
3.74	3.88	3.88	3.88	26.71	30.63	30.72	30.72	30.72	0.48
3.64	3.77	3.77	3.77	26.00	29.82	29.90	29.90	29.90	0.48
3.69	3.82	3.82	3.82	26.33	30.19	30.28	30.28	30.28	0.47
3.66	3.79	3.79	3.79	26.15	29.98	30.07	30.07	30.07	0.47
3.62	3.75	3.75	3.75	25.81	29.60	29.68	29.69	29.69	0.48
3.71	3.84	3.84	3.84	26.45	30.32	30.41	30.41	30.41	0.47
3.92	4.06	4.06	4.06	27.95	32.05	32.14	32.14	32.14	0.47
4.16	4.31	4.31	4.31	29.69	34.04	34.14	34.14	34.14	0.47
4.35	4.50	4.50	4.50	31.02	35.57	35.67	35.67	35.67	0.47
4.27	4.42	4.42	4.42	30.45	34.91	35.01	35.01	35.01	0.47
4.13	4.28	4.28	4.28	29.47	33.79	33.88	33.88	33.88	0.46
4.10	4.25	4.25	4.25	29.28	33.57	33.67	33.67	33.67	0.47
4.08	4.22	4.22	4.22	29.10	33.37	33.46	33.46	33.46	0.47
4.02	4.16	4.16	4.16	28.69	32.89	32.99	32.99	32.99	0.47
3.95	4.09	4.09	4.09	28.18	32.31	32.40	32.40	32.40	0.48
3.87	4.01	4.01	4.01	27.65	31.70	31.79	31.80	31.80	0.48
3.94	4.08	4.08	4.08	28.15	32.28	32.37	32.37	32.37	0.48
4.01	4.15	4.15	4.15	28.62	32.82	32.91	32.91	32.91	0.48
4.16	4.31	4.31	4.31	29.70	34.05	34.15	34.15	34.15	0.48
4.21	4.36	4.36	4.36	30.05	34.45	34.55	34.56	34.56	0.48
4.22	4.37	4.37	4.37	30.10	34.52	34.62	34.62	34.62	0.48
4.07	4.21	4.21	4.21	29.04	33.30	33.39	33.39	33.40	0.48
3.81	3.94	3.94	3.94	27.17	31.15	31.24	31.24	31.24	0.49
3.66	3.79	3.79	3.79	26.10	29.92	30.01	30.01	30.01	0.49
3.74	3.87	3.87	3.87	26.70	30.61	30.70	30.70	30.70	0.49
3.95	4.09	4.09	4.09	28.18	32.31	32.41	32.41	32.41	0.49
4.14	4.28	4.28	4.28	29.53	33.86	33.96	33.96	33.96	0.49
4.21	4.36	4.36	4.36	30.03	34.43	34.53	34.53	34.53	0.49
4.02	4.16	4.16	4.16	28.70	32.90	33.00	33.00	33.00	0.49
3.94	4.09	4.09	4.09	28.16	32.28	32.38	32.38	32.38	0.49
...
```

paulvha, 2019-04-06:
Thanks for nice remark. I have added in the SPS30.odt (you can read this with MS-word or OpenOffice or Libreoffice) in de Appendix B sample output from different sketches for reference. I also use the ESP32 (among others) but make sure to power the SPS30 with the VUSB (5 V) as pointed out in the top of the different sketches. That said.. yes I do see the same (I have 2 SPS30 and both have the same behavior). Actually in chapter 7.4 of the same document I make the (personal) conclusion, after a number of test and compare to other sensors, that I wonder about the ability to detect /report correctly above 2.5um. I have reported that to Sensirion but no reaction. If you use example 9, it will display with higher precision and then it shows some differences.

regards,
Paul

jhinderer, 2019-04-06:
Thank you for the quick response.
I tried different power sources (usb, batteries with voltage regulator and power supply) but the behaviour is identical. I will try your example 9. Maybe it will help. Otherwise the ability of the sensor is questionable. I am using a SDS011 for a year now and come to a similar conclusion as in your document. the PM2.5 values of the SDS011 and SPS30 are pretty similiar but PM10 is way off.
Is your report to Sensirion public? I would like to follow it (if their is any reaction in the future).

paulvha, 2019-04-06:
hi
I had just dropped a mail to the help-desk of Sensirion. I had good contact before with them about some SCD30 issues. They did not have a proper library for the SPS30 (just a bunch of pieces of starting code to connect, which I felt was over-engineered and not providing a solution). Actually many people from around the world are using my library for the same reason.  Some good working happening here : https://community.particle.io/t/dust-sensor-pms-5003-6003-7003/24221/102

sranjha, 2019-05-02:
```
21:00:52.970 -> -------------Mass -----------    ------------- Number --------------   -Average-
21:00:52.970 ->      Concentration [μg/m3]             Concentration [#/cm3]             [μm]
21:00:52.970 -> P1.0	P2.5	P4.0	P10	P0.5	P1.0	P2.5	P4.0	P10	PartSize
21:00:52.970 -> 
21:00:52.970 -> 0.66	0.68	0.68	0.68	0.00	0.00	0.00	0.00	0.00	0.00
21:00:56.160 -> 1.55	3.10	4.30	4.63	0.00	0.00	0.00	0.00	0.00	0.00
21:00:59.347 -> 1.48	2.89	3.98	4.29	0.00	0.00	0.00	0.00	0.00	0.00
21:01:02.582 -> 1.36	2.59	3.53	3.80	0.00	0.00	0.00	0.00	0.00	0.00
21:01:05.769 -> 1.12	1.16	1.16	1.16	0.00	0.00	0.00	0.00	0.00	0.00
21:01:08.957 -> 1.17	1.66	2.01	2.11	0.00	0.00	0.00	0.00	0.00	0.00
21:01:12.191 -> 1.24	1.62	1.88	1.95	0.00	0.00	0.00	0.00	0.00	0.00
21:01:15.378 -> 1.45	1.77	1.98	2.04	0.00	0.00	0.00	0.00	0.00	0.00
21:01:18.566 -> 1.59	1.94	2.17	2.23	0.00	0.00	0.00	0.00	0.00	0.00
21:01:21.789 -> 1.64	1.98	2.21	2.27	0.00	0.00	0.00	0.00	0.00	0.00
21:01:24.977 -> 1.60	1.93	2.16	2.22	0.00	0.00	0.00	0.00	0.00	0.00
21:01:28.211 -> 1.53	1.85	2.05	2.11	0.00	0.00	0.00	0.00	0.00	0.00
21:01:31.399 -> 1.44	1.70	1.86	1.91	0.00	0.00	0.00	0.00	0.00	0.00
21:01:34.586 -> 1.47	1.72	1.88	1.93	0.00	0.00	0.00	0.00	0.00	0.00
21:01:37.821 -> 1.52	1.76	1.91	1.95	0.00	0.00	0.00	0.00	0.00	0.00
21:01:41.008 -> 1.50	1.72	1.85	1.89	0.00	0.00	0.00	0.00	0.00	0.00
21:01:44.196 -> 1.46	1.68	1.81	1.85	0.00	0.00	0.00	0.00	0.00	0.00
21:01:47.430 -> 1.39	1.59	1.71	1.74	0.00	0.00	0.00	0.00	0.00	0.00
21:01:50.618 -> 1.34	1.52	1.62	1.65	0.00	0.00	0.00	0.00	0.00	0.00
21:01:53.805 -> 1.25	1.43	1.53	1.56	0.00	0.00	0.00	0.00	0.00	0.00
21:01:56.992 -> 1.18	1.35	1.46	1.49	0.00	0.00	0.00	0.00	0.00	0.00
21:02:00.227 -> 1.16	1.33	1.44	1.47	0.00	0.00	0.00	0.00	0.00	0.00
21:02:03.414 -> 1.10	1.25	1.34	1.37	0.00	0.00	0.00	0.00	0.00	0.00
21:02:06.649 -> 1.11	1.25	1.32	1.35	0.00	0.00	0.00	0.00	0.00	0.00
21:02:09.836 -> 1.14	1.30	1.38	1.41	0.00	0.00	0.00	0.00	0.00	0.00
21:02:13.024 -> 1.20	1.36	1.44	1.47	0.00	0.00	0.00	0.00	0.00	0.00
21:02:16.258 -> 1.25	1.40	1.49	1.52	0.00	0.00	0.00	0.00	0.00	0.00
21:02:19.430 -> 1.25	1.41	1.51	1.53	0.00	0.00	0.00	0.00	0.00	0.00
21:02:22.618 -> 1.20	1.35	1.43	1.46	0.00	0.00	0.00	0.00	0.00	0.00
21:02:25.852 -> 1.16	1.30	1.38	1.41	0.00	0.00	0.00	0.00	0.00	0.00
21:02:29.039 -> 501.07	537.22	550.88	554.75	0.00	0.00	0.00	0.00	0.00	0.00
21:02:32.274 -> 160.71	169.65	171.90	172.53	0.00	0.00	0.00	0.00	0.00	0.00
21:02:35.461 -> 33.95	35.55	35.79	35.86	0.00	0.00	0.00	0.00	0.00	0.00
21:02:38.649 -> 5.24	5.48	5.52	5.53	0.00	0.00	0.00	0.00	0.00	0.00
21:02:41.883 -> 2.11	2.21	2.23	2.23	0.00	0.00	0.00	0.00	0.00	0.00
21:02:45.071 -> 1.29	1.35	1.36	1.37	0.00	0.00	0.00	0.00	0.00	0.00
21:02:48.258 -> 1.75	1.83	1.84	1.85	0.00	0.00	0.00	0.00	0.00	0.00
21:02:51.492 -> 1.78	1.87	1.88	1.88	0.00	0.00	0.00	0.00	0.00	0.00
21:02:54.680 -> 2.20	2.31	2.33	2.33	0.00	0.00	0.00	0.00	0.00	0.00
21:02:57.867 -> 1.06	1.11	1.12	1.12	0.00	0.00	0.00	0.00	0.00	0.00
21:03:01.055 -> 1.31	1.37	1.38	1.39	0.00	0.00	0.00	0.00	0.00	0.00
21:03:04.289 -> 1.34	1.40	1.41	1.41	0.00	0.00	0.00	0.00	0.00	0.00
21:03:07.524 -> 1.40	1.46	1.46	1.46	0.00	0.00	0.00	0.00	0.00	0.00
21:03:10.711 -> 1.48	1.54	1.54	1.54	0.00	0.00	0.00	0.00	0.00	0.00
21:03:13.899 -> 1.48	1.54	1.54	1.54	0.00	0.00	0.00	0.00	0.00	0.00
21:03:17.086 -> 1.50	1.56	1.57	1.57	0.00	0.00	0.00	0.00	0.00	0.00
21:03:20.296 -> 1.60	1.67	1.67	1.68	0.00	0.00	0.00	0.00	0.00	0.00
21:03:23.530 -> 1.83	1.91	1.91	1.91	0.00	0.00	0.00	0.00	0.00	0.00
21:03:26.718 -> 2.01	2.09	2.09	2.09	0.00	0.00	0.00	0.00	0.00	0.00
21:03:29.905 -> 2.18	2.27	2.27	2.27	0.00	0.00	0.00	0.00	0.00	0.00
```

I'm actually seeing a similar pattern. The spike in the values is from lighting a match. I'm using UART. Have you been able to find out what causes the identical values already?

winkj, 2019-05-02:
Regarding the identical values: From a cursory look over the code, I believe the values printed out are the values received from the sensor. The datasheet definition of PMx outputs as "particles with a size smaller than “x” micrometers (e.g., PM2.5 = particles smaller than 2.5 μm)"; as such, for example PM4 also includes particles that are PM2.5 and PM1.0. 

If you're interested in a histogram like data - i.e. minimal measures particle up to PM1, >PM1 to PM2.5, >PM2.5 to PM4 etc. - you will have to subtract the smaller concentrations. For the last line of the output above ```21:03:29.905 -> 2.18	2.27	2.27	2.27``` that would mean that the PM1 concentration is 2.18, the PM2.5 is 0.09 (2.27-2.18), and PM4 and PM10 are 0 respectively.

paulvha, 2019-05-03:
@sranjha : looking at your print-out I am surprised to see zeros in number when connecting with UART. (I would have expected that in case of I2C with small receive buffer)  That said: NO I don't know what it is nearly the same. The numbers are coming directly from the SPS30. I can only expect that the sensor is actually working better on 2.5um and I doubt the larger um values. If you use example9 it will display more number behind decimal point and see some (very small) differences.

@jhinderer : yes you are right about the histogram. used that approach in example9 but with more numbers after the decimal point.

NarinLab, 2020-03-24:
Hello, first i want to say thanks for the author, this library is far much better than the official SPS30 that keep showing probe failed.

But i want to share my problem regarding weird read result. After doing data reading for about ten minutes, my SPS30 just sending me the same value like i show on the picture below:
[image]

any ideas why this happend? FYI, i'm using basic reading example 1

paulvha, 2020-03-24:
Thanks

weird issue.

Some questions:

Do you power the SPS30 with 5V ?
Is the fan turning ?
Do you connect UART or I2C ?
If I2C did you apply pull-resistors ?
What is the board you connected the SP30 to and how ?
What happens if you blow some air into the fan when the output is flat. Does the value change ?

Maybe you enable debug ( #define DEBUG 2) and make a copy of the values when the output is not changing (just a couple fo examples) and share that. I can then look whether at the information sent by the SPS30.

regards,
Paul

[Image of Yaktocat]<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcamo.githubusercontent.com%2Fdcfbffdca23f1f3c48cb0773ee1be070ea2a58ab%2F68747470733a2f2f692e6962622e636f2f767732665932372f436170747572652e6a7067&data=02%7C01%7C%7Cb98411ed563b46b659a308d7d003dd78%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637206587321474349&sdata=u1ipS9nbxuh3tiOeatTNQuTuUXdFvhhCmXMBpF4enI8%3D&reserved=0>

NarinLab, 2020-03-24:
Do you power the SPS30 with 5V ? 
**5Volt and stable power confirmed using Fluke multitester**

Is the fan turning ? 
**Fan is running about 2 second while measuring and data is visible from serial.print**

Do you connect UART or I2C ? 
**By UART**

If I2C did you apply pull-resistors ? 
**No I2C**

What is the board you connected the SP30 to and how ? 
**Arduino Mega2560**

What happens if you blow some air into the fan when the output is flat. Does the value change ? **Still experimenting to reproduce the weird constant result...**

paulvha, 2020-03-24:
I have tested on  the Mega2560 succesfully and that should not be an issue.

with respect to : Is the fan turning ?
Fan is running about 2 second while measuring and data is visible from serial.print

Did you change something ? e.g. included a sps30.stop() in the loop().  Normally the fan keeps on running and does not stop.

regards,
Paul

NarinLab, 2020-03-24:
Yes, the fan is running for two second, because I put sps30.stop() right after getting the sensor value. Is that a problem? Or can you give me some good pseudo code on the best method to doing measurement, most safe and most efficient.
Thankyou

Btw, now i just move my device to my living room, before (when the weird value incident happened) I put it outside, in front of my shop near the street. I got very fluctuative result:
[image] 
I'm just little bit frustrated, because when I put it back outside, after a while, it will be do the weird stuff again.

paulvha, 2020-03-24:
Maybe you should give it more time after sps30.start(), e.g. add a delay of say 10 seconds to allow enough airflow to happen before you read the values. if that helps, you can then always  try to reduce that time.  It might be a reason when you moved outside to have more airflow.  YOu could also remove the sps30.stop() and see what happens with your results.

Although, in theory, it COULD (potentially) also be because of sunlight interfering with the optical sensor (although I doubt that)

regards,
Paul

[Capture]<https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcamo.githubusercontent.com%2F36df8e88507e5a6af200dc1adf6f524d2f8f8507%2F68747470733a2f2f692e6962622e636f2f534a534b35324a2f436170747572652e6a7067&data=02%7C01%7C%7C2a5569445c8f4e9c65ba08d7d009aad4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637206612240328305&sdata=JDtANDl42EVzcelX8%2FCeert8khxhjp6oeSFmKyulCL8%3D&reserved=0>

NarinLab, 2020-03-24:
After doing more deep experiment, **the code: without delay and without stop and doing measurement every 3 second**; I checked the serial log and there is error message that state **Could NOT start measurement**. So, I can confirm this is the cause of weird constant value. The measurement is failed to start. When the device is outside, I unable to check ti serial log, now I see it, and still applying your suggestion regarding delay, hope this will resolve the proble.

Regards,
Narin

jwgmeligmeyling, 2020-03-24:
Maybe worthwhile to check out Sensirions own drivers:

I2C: https://github.com/Sensirion/embedded-sps
UART: https://github.com/Sensirion/embedded-uart-sps

Curious to try out on my own SPS30 sensors, after I had kind of abandoned them after just reading false values..

paulvha, 2020-03-24:
hi

Well that is progress. I had another person a couple of weeks ago who had issues with writting. Turned out that a small delay was needed between reading and writting. A variable RX_DELAY_MS is now in sps30.h for 200ms. Maybe increase that to 400 or 500ms just to see what happens

regards,
Paul

paulvha, 2020-03-24:
Thanks Jan-Willem

I had another look at the github. Some conclusion as before that these are the generic drivers for UART and I2C. Not really usable for Arduino-variant and missing examples.

The production of the SPS30 has moved from Swiss to China and while the quality should have remained the same over the past 2 months heared someone suffered from the need for longer time between read and write.  I assumed a specific SPS30 / component issue, but maybe it is wider. I have made adjustment to handle in the latest driver version.

Personally I have 2 SPS30 and never had any issues, nor heard any issues untill recently around that.

regards,
Paul

jhinderer, 2020-03-24:
Oh I totally forgot this issue.
But I still have problems to read different values for the particle sizes. No matter what I do they are always the same. At new years eve I had a slightly larger PM10 spike compared to the PM2.5. But it really was negliable (1607.44 µg/m³ and 1608.86 µg/m³).
I think that I got sensors from a faulty batch. Especially because in Narins diagram there are different values for the different particle sizes.

NarinLab, 2020-03-24:
Hi,

My sensor still stuck, and displaying a same value over and over. When i'm reset the Mega 2560, the sensor is updating new value... but after some minutes, then again... just repeating same value.

After checking log carefuly (appended it to my web interface via ESP32 Serialcomm, because physical Serial cable not possible for me in the ourdoor location), i just found the step by step why the sensor stuck at the same value:
My Web log:
```
25/Mar/20 05:31 -
25/Mar/20 05:32 -
25/Mar/20 05:36 -
25/Mar/20 05:40 - Failed to start
25/Mar/20 05:46 - Other error in reading: 81
25/Mar/20 05:50 - Failed to start
25/Mar/20 05:54 - Not found
25/Mar/20 05:59 - Other error in reading: 81
25/Mar/20 06:03 - Not found
```

My Code after some experiment with reset and close on error:
```
/* 
 *  @brief: Fungsi untuk get data SPS30
 *  @param: Struct references from DATA_SPS30 (modified)
 *  @return: Bool, true OK | false Err
 */
bool sensory_get_sps30(DATA_SPS30& data){
  SPS30 sps30;
  if (sps30.begin(SP30_COMMS) == false) {
    Log.fatal(F("SPS30 tidak dapat membaca sensor via Serial!" CR));
    data.logs = String(F("Serial failed"));
    return false;
  }
  if (sps30.probe() == false) {
    Log.fatal(F("SPS30 tidak ditemukan!" CR));
    data.logs = String(F("Not found"));
    return false;
  }
  if (sps30.reset() == false) {
    Log.error(F("SPS30 tidak dapat mereset!" CR));
    data.logs = String(F("Failed to reset"));
    return false;
  }
  if (sps30.start() == true) {
    Log.notice(F("SPS30 pengukuran dimulai!" CR));
  }
  else {
    Log.error(F("SPS30 tidak dapat memulai pengukuran!" CR));
    data.logs = String(F("Failed to start"));
    sps30.stop();
    sps30.reset();
    return false;
  }

  delay(30000);
 
  uint8_t ret, error_cnt = 0;
  struct sps_values val;

  // loop to get data
  do {

    ret = sps30.GetValues(&val);

    // data might not have been ready
    if (ret == ERR_DATALENGTH || val.MassPM1 == 0){

        if (error_cnt++ > 3) {
          Log.error(F("Error during reading values: %d" CR), (int)ret);
          data.logs = String(F("Error in reading: ")) + String(ret);
          sps30.stop();
          sps30.reset();
          return(false);
        }
        delay(1000);
    }
    // if other error
    else if(ret != ERR_OK) {
      Log.error(F("Error during reading values: %d" CR), (int)ret);
      data.logs = String(F("Other error in reading: ")) + String(ret);
      sps30.stop();
      sps30.reset();
      return(false);
    }

  } while (ret != ERR_OK);
  data.massPM1 = val.MassPM1;
  data.massPM2 = val.MassPM2;
  data.massPM4 = val.MassPM4;
  data.massPM10 = val.MassPM10;
  data.numPM0 = val.NumPM0;
  data.numPM1 = val.NumPM1;
  data.numPM2 = val.NumPM2;
  data.numPM4 = val.NumPM4;
  data.numPM10 = val.NumPM10;
  data.tps = val.PartSize;

  //delay(2000);
  if (sps30.stop() == true) {
    Log.notice(F("SPS30 pengukuran selesai!" CR));
  }
  return true;
}
```

Screenshot of the weird constant value after some interval (marked in red circle):
[image]
[image]
[image]

Any help would much appreciated!

paulvha, 2020-03-24:
hi

How does this program fit with Setup() and loop()?

#### global ONCE :

  SPS30 sps30;

#### functions that need to be called only during setup (once after start) :

 if (sps30.begin(SP30_COMMS) == false) {
    Log.fatal(F("SPS30 tidak dapat membaca sensor via Serial!" CR));
    data.logs = String(F("Serial failed"));
    return false;
  }
  if (sps30.probe() == false) {
    Log.fatal(F("SPS30 tidak ditemukan!" CR));
    data.logs = String(F("Not found"));
    return false;
  }
  if (sps30.reset() == false) {
    Log.error(F("SPS30 tidak dapat mereset!" CR));
    data.logs = String(F("Failed to reset"));
    return false;
  }

#### during loop () :

if (sps30.start() == true) {
    Log.notice(F("SPS30 pengukuran dimulai!" CR));
  }
  else {
    Log.error(F("SPS30 tidak dapat memulai pengukuran!" CR));
    data.logs = String(F("Failed to start"));
    sps30.stop();
    sps30.reset();
    return false;
  }

  delay(30000);

  uint8_t ret, error_cnt = 0;
  struct sps_values val;

  // loop to get data
  do {

    ret = sps30.GetValues(&val);

    // data might not have been ready
    if (ret == ERR_DATALENGTH || val.MassPM1 == 0){

        if (error_cnt++ > 3) {
          Log.error(F("Error during reading values: %d" CR), (int)ret);
          data.logs = String(F("Error in reading: ")) + String(ret);
          sps30.stop();
          sps30.reset();
          return(false);
        }
        delay(1000);
    }
    // if other error
    else if(ret != ERR_OK) {
      Log.error(F("Error during reading values: %d" CR), (int)ret);
      data.logs = String(F("Other error in reading: ")) + String(ret);
      sps30.stop();
      sps30.reset();
      return(false);
    }

  } while (ret != ERR_OK);
  data.massPM1 = val.MassPM1;
  data.massPM2 = val.MassPM2;
  data.massPM4 = val.MassPM4;
  data.massPM10 = val.MassPM10;
  data.numPM0 = val.NumPM0;
  data.numPM1 = val.NumPM1;
  data.numPM2 = val.NumPM2;
  data.numPM4 = val.NumPM4;
  data.numPM10 = val.NumPM10;
  data.tps = val.PartSize;

  //delay(2000);
  if (sps30.stop() == true) {
    Log.notice(F("SPS30 pengukuran selesai!" CR));
  }
  return true;
}

[SPS30 Constant Value]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcamo.githubusercontent.com%2Fab3768b9fc36d64e2bd127813b3e9feeebd3650b%2F68747470733a2f2f692e6962622e636f2f434836533050782f312e6a7067&data=02%7C01%7C%7Cff6123933bea4c152f2a08d7d0408659%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637206847850593021&sdata=eFJY0LHhdwnqi24bIOK5Y1RzlgrxCXqdt%2FxyUepP2N4%3D&reserved=0>
[SPS30 Constant Value]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcamo.githubusercontent.com%2Ffca65cb9ee8c7e7d25deb4259d9ddb94b9554b14%2F68747470733a2f2f692e6962622e636f2f647257714272782f322e6a7067&data=02%7C01%7C%7Cff6123933bea4c152f2a08d7d0408659%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637206847850593021&sdata=WttQ58rDf4KnBja%2Fd1zLJtI%2BaLjsrKFvinjgLY%2FVxEQ%3D&reserved=0>
[SPS30 Constant Value]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcamo.githubusercontent.com%2F1fd177163e7abd2a0d5d8553cf68cb8c11c931d9%2F68747470733a2f2f692e6962622e636f2f4c72743031506e2f332e6a7067&data=02%7C01%7C%7Cff6123933bea4c152f2a08d7d0408659%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637206847850603019&sdata=AgTGrzgeYWm9G61z%2BpL05vdOORok65Y4TwsJEZD7P8c%3D&reserved=0>

fboehle, 2020-03-25:
You might not have a faulty device. A research paper concluded that the Sensirion Sps30 is not able to detect particles above ~2 um:

https://www.researchgate.net/publication/337852062_Laboratory_evaluation_of_particle_size-selectivity_of_optical_low-cost_particulate_matter_sensors

> Oh I totally forgot this issue.
> But I still have problems to read different values for the particle sizes. No matter what I do they are always the same. At new years eve I had a slightly larger PM10 [...]

abrauchli, 2020-04-23:
@firefly-serenity I think our engineers added footnote 4 on page 2 of the datasheet

https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.6_Particulate_Matter/Datasheets/Sensirion_PM_Sensors_SPS30_Datasheet.pdf

> PM4 and PM10 output values are calculated based on distribution profile of all measured particles.

jhinderer, 2020-04-25:
@fboehle Interesting read, thank you!

@abrauchli Thanks, I guess this explains the behaviour pretty good.

I will close this issue now.

closed 2020-04-25 by jhinderer