paulvha/sps30 #5: SPS30 - Not A Teamplayer - Problems with other devices on I2C

detamend, 2019-06-12:
First the important stuff: Paul: I love your library! SENSIRION should hire you for the job you did here.

SYSTEM:
ESP8266 / WeMOS D1 Board
Arduino IDE 1.8.9
esp8266 Library V2.5.2

I2C Devices
SENSIRION SPS30
BME680
D1 OLED-Display Shield

I'm developing a Sensor-Reader/Data-Logger device, set up the SPS30 pretty much the way you did in the examples.

I read the sensor every two seconds, together with the BME680 and update the Display (via the u2g8 Library). And I get garbage like that:

[19-06-12_garbage.txt](https://github.com/paulvha/sps30/files/3282789/19-06-12_garbage.txt)
(Tab separated text file)

Interestingly, if I disable the routines, that write the Display - the SPS30 works fine:

[19-06-12_ok.txt](https://github.com/paulvha/sps30/files/3282812/19-06-12_ok.txt)

I tried to leave the SPS30 "time to breathe" by tick-tocking every second between reading the SPS30 on even seconds and writing the display on the odd ones - no success.

Tried the combination SPS30/BME680/Display unter the ESP8266-Lib 2.4.2 and it worked. Sadly, we can't go back, b'c some upcomming devices depend on the 2.5.2 Library - welcome to dependency hell.

One more thing: currently the devices are read/written to in that oder:

tick:
- SPS30
- BME680

tock:
- Display

If I change the order, so the BME680 gets read _before_ the SPS30, it screws up the measurement again.

I know: your library comes without warranty or help - if you got any idea what's going on there, it would be greatly appreciated.

paulvha, 2019-06-12:
HI Detlef

Thanks for reaching out.  Currently working full time on other project but some thoughts.

Hard to understand what is really happening. Weird that it works with 2.4.2 and not with 2.5.2…
I had created an example with a BME280 only/ The BME680 did not add enough change for me.

I also see GPS location in the documents. How was that added ?
Could there be any issue with clock stretch ?  Normally the BME680 does not need that, but still.
Can you share the .ino you use? Maybe I can spot something in there?
Do you have pull resistors on I2C ? The SPS30 is very sensitive that ?

Greetings from the Netherlands,

Kind regards

Paul

detamend, 2019-06-12:
Paul, I'm really grateful for your input!

BME680 <> BME280 - we wanted to have the option to use this one; and yeah, it's pretty similar. In the current devBranch a 280 would do fine.

GPS - a somple GPS Module, connected via Serial and processed by tinyGPS++

Clock Strech - I remember that I read about it somewhere, but can't tell right now what that does.

Pull Ups - check, 10k x 2

I cobbled together the routines that matter in this ZIP:

[SPS30_DataLog.zip](https://github.com/paulvha/sps30/files/3283190/SPS30_DataLog.zip)

It's a multi-Tabbed INO, so I put all the functions into different files to keep my head at least a bit over water.

May start with 03_LOOP.ino - that's the sequence, that loops through the different modules. From there you find the routines in the so named *.inos.

I left Debug, WLAN auf a bunch of stuff out - so you won't have to fight through these ;)

Thanks in advance!!!

zz-zsys, 2019-06-12:
Hi @detamend,

This problem has occured to me before. It tooks me a while to find out why this problem occured.
The reason why this happens is because of the U8G2 library that you used. The default configuration for the i2c clock for SSD1306 on that library is 400kHz. Meanwhile SPS30 only supports max. 100 kHz. BME680 or BME280 still function because they do still support up to 400 kHz. As a solutions, set your display i2c clock to 100 kHz using u8g2.setBusClock function before u8g2.begin function. Hope this will solved your problem.

Best regards from Berlin,
ZZ

detamend, 2019-06-12:
zz-zsys - you're my hero! I owe you guys a beer or two!

Thanks a lot!!

paulvha, 2019-06-13:
Great Teamwork !!!. The real power of open source and sharing 😊

Regards,
Paul

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

paulvha, 2019-06-13:
issue closed

closed 2019-06-13 by paulvha

Mohit2183, 2020-01-11:
we were testing your sps30 by connecting with docklight.  we have used usb to serial converter to command the sps30. pin 1 was connected to poer supply +5vdc , pin 5 was connected to 0vdc. pin4 was left open for UART, pin 2 was connected to Tx of master (PC), pin 3 was connected to Rx of master (pc). From docklight v2.3  we had set the com port , 115200, 8N1 communication settings.  we have placed command in format of Hex codes i.e. start command Tx: 7E 00 00 01 02 F9 7E , Rx REPLY: 7E 00 00 01 02 F9 7E , Tx: 7E 00 00 00 FF 7E, Rx REPLY: x: 7E 00 00 00 FF 7E , Tx: 7E 00 03 00 FC 7E ,  Rx REPLY: 7E 00 03 00 FC 7E , Tx:  7E 00 D3 00 2C 7E, Rx REPLY:  7E 00 D3 00 2C 7E .   OVER ALL FAN IS ALSO NOT ROTATING PHYSICALLY.. could you pls guide us why sps30 is not giving response to our commands on dock light.