paulvha/sps30 #6: Setting up UART

yukimach, 2019-06-26:
Hi, I'm having some trouble getting the code running on an ESP32 via UART. I've commented out all the I2C parts and have it compiling. when I turn the debug mode on, I see that the data being sent but no reply. I've got the wires hooked up as in the document, and have turned Serial2 on for the pin 17 and 16 serial2. Could you advise on how I can debug the issue? Thanks!!

paulvha, 2019-06-27:
Hi

Pin 16 and 17 are the default for Serial2. I assume you have selected : #define SP30_COMMS SERIALPORT2 (around line 150)

Check that RX and TX line are connected correctly (maybe exchange for a test). On the cable I have (left to right when looking at the back of the SPS30):

SPS30 COLOUR  ESP32

VCC     red          VUSB
RX       white       GPIO17
TX       purple      GPIO16
select  green       Not connected
gnd      black       GND

If that does not work try to use Serial1 for a test :#define SP30_COMMS SERIALPORT1 (around line 150)

Connect and select pins for TX and RX (around line 158)
#define TX_PIN 26
#define RX_PIN 25

SPS30 COLOUR  ESP32

VCC     red          VUSB
RX       white       GPIO26
TX       purple      GPIO25
select  green       Not connected
gnd      black       GND

P.s. There should have been no need to remove all the I2C code. See chapter 3.3.2 Exclude I2C communication code in the ODT-document (can be opened with nearly any word or office package).

regards,
Paul

yukimach, 2019-06-27:
Yup ok it works. Had a small wiring issue. Thank you for the rapid response!!

paulvha, 2019-06-27:
Good to hear... enjoy...

Regards
Paul

Outlook voor Android downloaden<https://aka.ms/ghei36>

closed 2019-06-27 by yukimach