paulvha/sps30 #7: could not probe / connect with SPS30.

yukimach, 2019-11-27:
Hi, 

Does the fan turn on no matter what? It was working, and tried to upload new code and now it doesnt work. I hadn't touched code in a while so something may have changed.. I need to debug this.  I get: 

Trying to connect
could not probe / connect with SPS30.
Program on hold

Should the fan turn on once plugged in, or is it after the mcu communicates with the sps? 
I'm tripple checking the rx tx pins. I'm setting Serial1 and defining RX TX as I'm using an ESP32. 
On pin 17 16.

paulvha, 2019-11-27:
HI

The fan will not turn on until you start measurement.  The error is very basic.. it does not get a response from the SPS30

How did you connect all the 5 wires?
What interface did you select in the sketch  for the SPS30?

Regards,
Paul

yukimach, 2019-11-27:
Hi, so I connected
  SPS30 pin     External     ESP32
 *  1 VCC -------- 5V
 *  2 TX -----------------------RX 17
 *  3 RX -----------------------TX 16
 *  4 Select ----- Not Connected
 *  5 GND -------- GND --------- GND

I've tried with SERIALPORT1 and SERIALPORT2 also tried setting the pin rx tx to 0.

paulvha, 2019-11-27:
Hi

The 5V is that external or did you use VUSB on the ESP32 ?
Try pin 26 and 25 with  SERIALPORT2

Regards,
Paul

yukimach, 2019-11-27:
Will try .. but Its a custom board now and yes supplied from 5v usb. 
It was definitely functioning previous to updating the code just today..

paulvha, 2019-11-27:
If it worked before on Serial1 did you change ???

/////////////////////////////////////////////////////////////

/* define RX and TX pin for softserial and Serial1 on ESP32

* can be set to zero if not applicable / needed           */

/////////////////////////////////////////////////////////////

#define TX_PIN 26

#define RX_PIN 25

yukimach, 2019-11-27:
Yes I changed that part. I'll retry it again on a feather huzzah this evening.

yukimach, 2019-11-27:
Trying to connect
Sending:  0x7E 0x00 0xD0 0x01 0x03 0x2B 0x7E
TimeOut during reading byte 0

I should be receiving a reply from the sps right? so the communication is not working?

paulvha, 2019-11-27:
HI

This is the first step : sending a request for the firmware and indeed should be answered by the SPS30. Often RX and TX are exhanged, just try it (it will not harm 🙂).

If not working still , sent me the sketch and your hardware setup on paulvha@hotmail.com. I will have a look at it.

regards,
Paul

yukimach, 2019-12-02:
Thanks, I might order another sensor just incase its that.
Could it be that my espsoftwareserial isn't the right library? its hardwareserial so this shouldn't matter right?

paulvha, 2019-12-02:
Indeed hardware serial has no impact from softserial. Still sent the sketch as you use it and the hardware connections.  It might be a small thing that is easy to overlook

Regards
Paul

Outlook voor Android downloaden<https://aka.ms/ghei36>

paulvha, 2019-12-02:
Fixed in version 1.3.7

closed 2019-12-02 by paulvha

silviupanaite, 2021-04-04:
Hi, I'm having the same issue (could not probe/connect with SPS30.). I have tried i2c on ESP8266E and serial on ESP32. Is there a way to test to see if the SPS is damaged or my doggy wiring?

paulvha, 2021-04-05:
hi,
Maybe it is broken.. but maybe it is another issue.

How did you connect the wiring?
What sketch do you use?

regards,
Paul

silviupanaite, 2021-04-05:
On ESP8266 (NodeMCU 0.9) I used the Example13_sps30_BasicReadings_any_I2C.ino with D1, D2 pins and pull-up resistor 10K and also tried with 4k7 using external HW-131 Breadboard Power Module 3.3V/5V witch also powers the module, I used the diagram from https://www.14core.com/wp-content/uploads/2019/09/senserion-sps30-particle-sensor-schematics-diagram.jpg. For the serial, I tried on Wemos Lolin32 Lite pin (16,17) and (25,26) (switching the pairs around) without the pull-up resistors.

I managed to get some readings once without the pull-up resistors but I was fiddling with the SPS30 power connector (I was using the Arduino-SPS library), I also tested the cables for continuity and V and found no issues. Is there anything more I can try?

paulvha, 2021-04-05:
I don't have a NodeMCU 09, but see that D1 is CLOCK, D2 SDA. However on other drawings I see they D1 is SDA and D2 is SCL. Have you tried to switch the SDA /SCL line just for debug?
Have you connected other I2C devices before and did that work?
is you I2C init something like Wire.begin(1, 2); // sda, scl ?

Doing a pull-up to 5V is not a good idea, it should be to 3v3. The SPS30 absulutely needs pull-up to work.

regards,
Paul

silviupanaite, 2021-04-05:
I don't have experience with I2C. But thins are looking stranger and stranger, I connected the NodeMCU 3.3v and the power supply 3.3V on the same side from which the pull-up resistors are connected... and the library managed to: Detected SPS30. -> Serial number: CE7A73B4FFA7A053 -> Product name: 00080000 -> Can not read version info.

Now the **StatusCheck()** returns **false**.  (ret = sps30.GetStatusReg(&st); Serial.println(ret); witch is 136);

If I bypass that (  if (st == 0) return(true);) it showed some readings at the beginning, but now they're showing 0.0.

paulvha, 2021-04-05:
hi

THis is progress...

The SPS30 really needs 5V to work on VCC else it is unstable.  The I2C pull-up resistors must be to 3v3. I have experienced with other 3V3 boards that input stopped as the voltage was too high (like an internal protection).

Looking at what you got that data is correct and thus SCL and SDA are connected correctly.

I wonder about your 5V on SPS30, can you double check that ?
When you tried serial, did you remove the pull-up resistors ? (next to removing the sense-wire)

regards,
Paul

silviupanaite, 2021-04-11:
On the serial, I did remove the pull-up resistors but it didn't work I will work on the I2C for now. Thank you for the support.

rictolcrow, 2021-07-09:
Hi. Was there any definitive solution to this problem? I have SPS30 connected to Serial1 UART on Feather M0 WiFi that has intermittent connection; sometimes it connects, other times it reports "could not probe / connect with SPS30." 
I also have in the system: BME280 (i2c), SCD30 (i2c), featherwing SD-RTC (SPI and i2c), and an Adafruit Miniboost to get the 3.7V to 5V. The SPS30 seems ok if I run it on a plain FeatherM0 with nothing else. I've tried multiple components and behavior is the same. Any ideas? Please, I'm pulling out my beard over this one.

paulvha, 2021-07-10:
I suspect the power supply,  especially as it runs well in standalone. Not sure whether you power from a lipo but if all the sensors  en WIFI do "something" at the same time it can draw too much current and cause a voltage drop.  Maybe just for debug, use a strong 5V, e.g. from the USB pin if connected for test to SPS30. 

regards
Paul

rictolcrow, 2021-07-10:
Thanks Paul. I also suspected power. I've run it from the USB power and an independent supply with the same intermittent results.  Interestingly, it seems to consistently connect to the SPS30 right after connecting the USB and turning on by de-grounding the ENable pin on the Feather's 3.3v regulator (a switch on the Featherwing Terminal block). However, if I just reset the Feather, it won't connect. Even if I turn off and on, it won't work. I have to do the whole magic pass of turning off, disconnecting the USB, reconnecting USB, and turning on to get a successful connection to the SPS30. I'll get an o'scope to sniff the RX/TX but I don't know what I'm looking for. More later...

paulvha, 2021-07-11:
maybe the SPS30 enable wire, which switches the protocol between Serial and I2C has some impact. The SPS30 reads this wire at power-up to determine its protocol. When not connected/floating, it uses Serial. Maybe, somehow, it gets confused and moves to I2C due to a voltage drop/spike.

rictolcrow, 2021-07-15:
I figured a workaround. It turns out that the SPS30 needs to restart in order to connect. If it isn't powered down the UART will not respond to a new connection. This I don't know why. 
When I was running it from USB power, disconnecting the USB cable effectively depowered it and restarted it and it would reconnect ok. When I was running it off the 5V from the Miniboost it never got de-powered because the Miniboost has a large capacitor and when turned off its output voltage drops slowly. Hence, when I'd hit reset on the Feather M0 it never would reset the SPS30!
The workaround is to use the ENable pin on the Miniboost. In void setup() I pull this pin to LOW for 10 secs to allow the 5V to drop to ~0.5V and then HIGH to get the 5V for the SPS30 and restart/reset it.