paulvha/sps30 #8: auto-clean on esp8266 not working

HyperDevil, 2020-02-12:
10:53:26.255 -> Serial number : 79D40290B4DFxxxx
10:53:26.255 -> Product name : not available
10:53:26.255 -> Article code : not available
10:53:26.255 -> could not get clean interval.Protocol error
10:53:26.255 -> No Auto Clean interval change requested.
10:53:27.251 -> Measurement started
10:53:27.251 -> 
10:53:27.251 -> Hit <enter> to continue reading
10:53:29.263 -> Hit <enter> to continue reading
10:53:31.248 -> Hit <enter> to continue reading
10:53:33.259 -> -------------Mass -----------    ------------- Number --------------   -Average-
10:53:33.259 ->      Concentration [μg/m3]             Concentration [#/cm3]             [μm]
10:53:33.293 -> P1.0	P2.5	P4.0	P10	P0.5	P1.0	P2.5	P4.0	P10	PartSize
10:53:33.293 -> 
10:53:33.293 -> 0.74	2.94	4.69	5.04	0.40	3.55	5.73	6.15	6.21	1.63
10:53:36.299 -> 0.79	1.49	2.01	2.12	3.92	5.58	6.26	6.39	6.41	0.89
10:53:39.297 -> 0.87	1.52	2.01	2.11	4.58	6.29	6.93	7.05	7.07	0.92

Using I2c on ESP8266.

paulvha, 2020-02-12:
HI

Have just compiled and checked example 2 to be sure. All works fine.

Do you have pull-up resistors on both SCL and SDA to 3V3?
Do you power the SPS30 with 5V external ?
the xxx behind the serial number did you add those manual ?

regards,
Paul

10:53:27.251 -> Hit to continue reading
10:53:29.263 -> Hit to continue reading
10:53:31.248 -> Hit to continue reading

HyperDevil, 2020-02-12:
Hi Paul,

Yes the pullups are there 10K to 3v3.
I use the VIN pin on the ESP8266 which is 5V (from usb), my USB port can supply 3A. I checked with a multimeter.
And yes i added the xxx (although it is not sensitive information).

HyperDevil, 2020-02-12:
`11:13:38.032 -> SPS30-Example2: Basic reading + clean. press <enter> to start
11:13:40.012 -> SPS30-Example2: Basic reading + clean. press <enter> to start
11:13:42.021 -> Trying to connect
11:13:42.021 -> I2C Sending:  0xD0 0x33
11:13:42.021 -> I2C Received: 0x37 0x39 0x44 0x34 0x30 0x32 0x39 0x30 0x42 0x34 0x44 0x46 0x33 0x33 0x33 0x37 0x00 0x00 length: 18
11:13:42.055 -> 
11:13:42.055 -> Detected SPS30.
11:13:42.055 -> I2C Sending:  0xD3 0x04
11:13:44.031 -> I2C Sending:  0xD0 0x33
11:13:44.031 -> I2C Received: 0x37 0x39 0x44 0x34 0x30 0x32 0x39 0x30 0x42 0x34 0x44 0x46 0x33 0x33 0x33 0x37 0x00 0x00 length: 18
11:13:44.065 -> 
11:13:44.065 -> Serial number : 79D40290B4DF3337
11:13:44.065 -> Product name : not available
11:13:44.065 -> I2C Sending:  0xD0 0x25
11:13:44.065 -> I2C Received: 0x00 0x00 length: 2
11:13:44.065 -> 
11:13:44.065 -> Article code : not available
11:13:44.065 -> I2C Sending:  0x80 0x04
11:13:44.065 -> Error: Received NO bytes
11:13:44.065 -> I2C Received: length: 0
11:13:44.065 -> 
11:13:44.065 -> Error during reading from I2C: 0x51
11:13:44.065 -> could not get clean interval.Protocol error
11:13:44.065 -> No Auto Clean interval change requested.
11:13:44.098 -> I2C Sending:  0x00 0x10 0x03 0x00 0xAC
11:13:45.085 -> Measurement started
`

paulvha, 2020-02-12:
It is clear the connection is working as you get the serial number and can read the results. I have just connected the ESP8266 and did the same test example2.  and added some comments in between. My SPS30 answers to  the request: I2C Sending:  0x80 0x04, where yours is not.

Try this : remove both power source, count to 10 and reconnect (5V first). Maybe the SPS30 is in state of flux.
What happens if you try to set the autocleaning ?

regards,
Paul

SPS30-Example2: Basic reading + clean. press <enter> to start
Trying to connect

I2C Sending:  0xD0 0x33
I2C Received: 0x33 0x36 0x31 0x44 0x34 0x46 0x44 0x33 0x41 0x41 0x39 0x34 0x31 0x41 0x37 0x36 0x00 0x00 length: 18
PROBE = SAME AS ASKING SERIAL NUMBER

Detected SPS30.

I2C Sending:  0xD3 0x04
RESET

I2C Sending:  0xD0 0x33
I2C Received: 0x33 0x36 0x31 0x44 0x34 0x46 0x44 0x33 0x41 0x41 0x39 0x34 0x31 0x41 0x37 0x36 0x00 0x00 length: 18
Serial number : 361D4FD3AA941A76

Product name : not available
NEVER AVAILABLE ON I2C

I2C Sending:  0xD0 0x25
I2C Received: 0x00 0x00 length: 2
Article code : not available

I2C Sending:  0x80 0x04
I2C Received: 0x00 0x09 0x3A 0x80 length: 4
Current Auto Clean interval: 604800 seconds

I2C Sending:  0x80 0x04 0x00 0x00 0x81 0x02 0x30 0x9D
I2C Sending:  0xD3 0x04
RESET
Auto Clean interval now set : 560 seconds

I2C Sending:  0x80 0x04
I2C Received: 0x00 0x00 0x02 0x30 length: 4
Current Auto Clean interval: 560 seconds

I2C Sending:  0x00 0x10 0x03 0x00 0xAC
Measurement started

HyperDevil, 2020-02-12:
Ok i did that now:

#define AUTOCLEANINTERVAL 604800
#define PERFORMCLEANNOW 0
#define DEBUG 2

11:49:10.725 -> SPS30-Example2: Basic reading + clean. press <enter> to start
11:49:12.727 -> Trying to connect
11:49:12.727 -> I2C Sending:  0xD0 0x33
11:49:12.727 -> I2C Received: 0x37 0x39 0x44 0x34 0x30 0x32 0x39 0x30 0x42 0x34 0x44 0x46 0x33 0x33 0x33 0x37 0x00 0x00 length: 18
11:49:12.760 -> 
11:49:12.760 -> Detected SPS30.
11:49:12.760 -> I2C Sending:  0xD3 0x04
11:49:14.729 -> I2C Sending:  0xD0 0x33
11:49:14.763 -> I2C Received: 0x37 0x39 0x44 0x34 0x30 0x32 0x39 0x30 0x42 0x34 0x44 0x46 0x33 0x33 0x33 0x37 0x00 0x00 length: 18
11:49:14.763 -> 
11:49:14.763 -> Serial number : 79D40290B4DF3337
11:49:14.763 -> Product name : not available
11:49:14.763 -> I2C Sending:  0xD0 0x25
11:49:14.763 -> I2C Received: 0x00 0x00 length: 2
11:49:14.763 -> 
11:49:14.763 -> Article code : not available
11:49:14.763 -> I2C Sending:  0x80 0x04
11:49:14.763 -> Error: Received NO bytes
11:49:14.763 -> I2C Received: length: 0
11:49:14.763 -> 
11:49:14.763 -> Error during reading from I2C: 0x51
11:49:14.797 -> could not get clean interval.Protocol error
11:49:14.797 -> I2C Sending:  0x80 0x04 0x00 0x09 0x09 0x3A 0x80 0xA7
11:49:15.783 -> I2C Sending:  0xD3 0x04
11:49:17.791 -> Auto Clean interval now set : 604800 seconds
11:49:17.791 -> I2C Sending:  0x80 0x04
11:49:17.791 -> Error: Received NO bytes
11:49:17.791 -> I2C Received: length: 0
11:49:17.791 -> 
11:49:17.791 -> Error during reading from I2C: 0x51
11:49:17.791 -> could not get clean interval.Protocol error
11:49:17.791 -> I2C Sending:  0x00 0x10 0x03 0x00 0xAC
11:49:18.777 -> Measurement started
11:49:18.777 -> 
11:49:18.777 -> Hit <enter> to continue reading
11:49:20.784 -> Hit <enter> to continue reading
11:49:22.790 -> Hit <enter> to continue reading
11:49:24.799 -> Hit <enter> to continue reading
11:49:26.803 -> Hit <enter> to continue reading

HyperDevil, 2020-02-12:
As discussed by email, add this to the sps30.cpp file on line 1144 ofter I2C_Setpointer() function.
delay(5);

closed 2020-02-12 by HyperDevil