paulvha/sps30 #12: SPS_DEBUGSERIAL_SODAQ Error

webseitz, 2020-04-05:
Dear Paul

I'm a newbie to Arduino and your package, so I'm not sure if the following error is due to a programming error on my side or a bug in the package.

I use a ATMega328P on a Chinese Arduino Uno (which worked well for other applications) wired according your sketch to the SPS30.
Now I wanted to upload/compile the example sketch number 1 from your package. I changed the lines to
#define SP30_COMMS I2C_COMMS
#define TX_PIN 0
#define RX_PIN 0
to communicate over I2C to the SPS. I receive the error message below. Since you added recently the feature DEBUGSERIAL I wondered if this is maybe a bug...

```
In file included from /Users/webseitz/Documents/Arduino/libraries/sps30-master/src/sps30.cpp:66:0:
/Users/webseitz/Documents/Arduino/libraries/sps30-master/src/sps30.cpp: In member function 'void SPS30::DebugPrintf(const char*, ...)':
/Users/webseitz/Documents/Arduino/libraries/sps30-master/src/sps30.h:72:33: error: 'SerialUSB' was not declared in this scope
 #define SPS30_DEBUGSERIAL_SODAQ SerialUSB
                                 ^
/Users/webseitz/Documents/Arduino/libraries/sps30-master/src/sps30.cpp:129:13: note: in expansion of macro 'SPS30_DEBUGSERIAL_SODAQ'
             SPS30_DEBUGSERIAL_SODAQ.print(prfbuf);
             ^~~~~~~~~~~~~~~~~~~~~~~
/Users/webseitz/Documents/Arduino/libraries/sps30-master/src/sps30.h:72:33: note: suggested alternative: 'Serial_RX'
 #define SPS30_DEBUGSERIAL_SODAQ SerialUSB
                                 ^
/Users/webseitz/Documents/Arduino/libraries/sps30-master/src/sps30.cpp:129:13: note: in expansion of macro 'SPS30_DEBUGSERIAL_SODAQ'
             SPS30_DEBUGSERIAL_SODAQ.print(prfbuf);
             ^~~~~~~~~~~~~~~~~~~~~~~
```

paulvha, 2020-04-05:
hi
Good catch !!
Nice of you to reach out. I have made an update yesterday (after testing on the SODAQ board) and realize now that I should have done more testing on other boards after making this change. Indeed line 129 in CPP should read SPS30_DEBUGSERIAL.print(prfbuf). I need to do better work for the sps30.h  (SerialUSB is defined for SODAQ, but not for another board) I'll post an update this evening.

Stay safe

regards,
Paul

paulvha, 2020-04-05:
updated!

closed 2020-04-05 by paulvha