paulvha/sps30 #14: Setting up serialport2 TX and RX

paulhanga, 2020-05-27:
Hi,
I'm having trouble to connect an sps30 to a TTGO wrover esp32. 
For serial port 2 I have to use 16,17 pins but the TTGO esp32 doesn't have this pinouts.
I tried to change them in the sketch with 4,5 but without success.
Have u got any hint?

Thnks 
Paul
anyway thanks for the lib, nice work

paulvha, 2020-05-27:
hi
I am applying a change to make this easier. Final tests and documentation update are being done, will be posted this evening.

regards,
Paul

paulhanga, 2020-05-27:
Hi,

that's wonderful.
Thank you very much.

Good work 

paul

paulvha, 2020-05-27:
new version has been uploaded. look at example12 to select your serial port of choice.

closed 2020-05-27 by paulvha

paulhanga, 2020-05-27:
Thank you very much for your help and great work. And thank you for sharing. 
Bye
--
Sent from myMail for Android Wednesday,

paulhanga, 2020-06-03:
Hi,

It works great.

Thanks again

Closed #14 [https://github.com/paulvha/sps30/issues/14].