paulvha/sps30 #16: ByteToFloat

matthijsberg, 2020-11-05:
Hi Paul, 

I tried compiling and example 7 this evening, but I run into those errors:

```
{
	"resource": "/Users/matthijsvandenberg/Documents/PlatformIO/Projects/AirQual Plafond/.pio/libdeps/d1_mini/sps30/src/sps30.h",
	"owner": "cpp",
	"severity": 8,
	"message": "conflicting declaration 'typedef union ByteToFloat ByteToFloat'",
	"startLineNumber": 329,
	"startColumn": 3,
	"endLineNumber": 329,
	"endColumn": 3
}
```
and
```
{
	"resource": "/Users/<naam>/Documents/PlatformIO/Projects/AirQual Plafond/lib/scd30/src/paulvha_SCD30.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'ByteToFloat' has a previous declaration as 'typedef union ByteToFloat ByteToFloat'",
	"startLineNumber": 110,
	"startColumn": 3,
	"endLineNumber": 110,
	"endColumn": 3
}
```

I try this with PlatformIO in VSCode. Any idea's? Google comes up empty. Is this an old version of something? Bit of a noob as you know. :-(  

Thnx.

paulvha, 2020-11-06:
hi Matthijs

The issue is because you have ByteToFloat double declared in your project. The complete project folder is compiled and it finds 2 times a declaration with the same name (does not matter whether the declaration is the same..)

You havea file :

Plafond/.pio/libdeps/d1_mini/sps30/src/sps30.h",

And you have a different file:

Plafond/lib/scd30/src/paulvha_SCD30.h

Indeed what I have done is use code, previous developed for SPS30, into my SCD30 library. Don't like to rewrite the same code twice. I did not consider anyone would use both SPS30 and SCD30 at the same time. I have just made a small change and uploaded a new version of the SCD30 ( no change to SPS30) to prevent this error happening again.

regards,
Paul

matthijsberg, 2020-11-06:
update, compiled the example on Arduino IDE and works instantly. Must be something specific to VS Code or PIO of the person behind the keyboard.

closed 2020-11-06 by matthijsberg