paulvha/sps30 #17: Hard reset?

uutzinger, 2020-12-10:
Is it possible to conduct a hard reset on the SPS30?
When uploading new code while a previous program is running and accessing the sensor through I2C, the command ".probe()" usually fails. If the sensor is powered down and reenergized, the program works as expected.

This likely is a firmware issue or perhaps when using I2C one needs to send reset before executing probe()?

paulvha, 2020-12-10:
hi Urs,

Strange.. I have not had that issue before.

What is the sketch you are running ?

Couple of options jumps to mind:

Are you using the sleep() option for the SPS30 ? If so, it might have been put to sleep and then needs a special sequence to wakup() (or  reset). Calling wakeup() when new code is loading will not help as the current code will check whether you had put it to sleep first ?

Normally at begin() the wire is reset, but not the SPS30, You could try to do a reset(), but I have not seen or heard the need for this before..

regards,
Paul

uutzinger, 2020-12-11:
That is interesting to hear. My firmware does not allow sleep.

But since I have many sensors on the same i2c bus, its possible that a software reboot leaves their state undetermined and sensors affect each other.
I will try with SPS30 being only sensor accessed. If that does not work I will remove physically other sensors from my board.

What I understand from your response is that under normal operation where sensor is only device on i2c bus, your example programs can be programmed to ESP8266 and one does not need power to down the  SPS30 to make them start properly.
Thank you for your response.

Urs Utzinger

External Email

paulvha, 2020-12-12:
HI Urs

It could indeed be that some sensors are in different state. Say a slave device is half way sending a reply or data, it may pull SDA low waiting for the next clock on SCL. If that never comes, it keeps the I2C bus locked and keep the bus locked low. Some Wire libraries will test at init whether SDA and SCL lines are responding as expected before returning OK.

There are ways to unlock an I2C bus/reset:

  *   If the data line (SDA) is stuck LOW, the master should send nine clock pulses. The device that held the bus LOW should release it sometime within those nine clocks. If not, then use the HW reset or cycle power to clear the bus. Depending on your board / Wire-library this hard reset might be available. Which board/library do you use ?
  *   When you sent on a wire to address 0x0 (zero) a  command  0x6 it might reset for (nearly) ALL the devices on the wire.

With the SPS30 I never had any problems to restart/reconnect/probe after uploading a new sketch update.

regards,
Paul

uutzinger, 2020-12-14:
Paul

thank you very much for your feedback. 
I went back to your example programs (read basics) and indeed the sps30 and system behaves as expected. 

I found that the LCD backpack (20x4 characters) causes the behavior I described above. Once I disable its initialization my program no longer hangs at the probe() command.  I replaced the LCD backpack with one from Adafruit and its corresponding LiquidCrystal library and it also causes same issue. I also noticed the MLX90614 sensor causing similar issue.

My interpretation is that the issue is within the i2c core of ESP8266 or the Lolin D1 Board I am using.

I will try to implement the bus reset when probe fails as you suggested. 
Please feel free to close this issue as I don't think this has anything to do with your library.

Urs

paulvha, 2020-12-14:
Thanks for the update

I looked at the source, the ESP8266 Wire implementation does not perform a bus-check at the start and does not have a bus-reset call.

Just out of interest.. do you have pull-up resistors on SDA and SCL and what value are they ?

regards,
Paul

uutzinger, 2020-12-14:
Thanks Paul,
I used level shifter breakout board 3.3 to 5V. The LCD backpack and the SPS30 are attached to the level shifter.
I don't recall the size of the pullups in the level shifter breakout board and whether its one from Sparkfun (Sparkfun uses 10k).
I will measure later today.
I will also check  on the 3.3 V side perhaps some of the breakout boards together add pullups in parallel.

uutzinger, 2020-12-15:
On 5V side, the level shifter has 5k pullup. The lcd backpack 10k. Total R 3.3k.
On 3.3V side its a mix of 5k and 10k on the breakout boards.  Total 1.3k.
I think I will need to use separate i2c bus for LCD backpack.

paulvha, 2020-12-15:
HI Urs

1.3K is on the low side indeed.. Normally I try to keep it 4K7 to 10K. It could cause the issue with draining too much power. If you have a scope you could detect it. On other aspect to consider is the total power consumption. A backlight on an LCD could easily take 200mA and then when reset it can cause a dip.

regards,
Paul

ra1nb0w, 2021-08-13:
Hi,
I am using the serial port with ESP32 with `sleep()`/`wakeup()` and each time I flash a new firmware either via USB or OTA the sensor is not probed. Just to confirm the library code I tried, in case of `probe()` fails, to `reset()`/`stop()`/`wakeup()`/ecc. and `esp_restart()` without any success. Do you have some suggestions on how to avoid this annoying problem and reinitialise the SPS30?

thank you very much for the hard work that you put in this library!

ra1nb0w, 2021-08-14:
for the moment, I will use `start()` and `stop()`

paulvha, 2021-08-18:
Hi,

I am on a short break and will look at this when I am back.

Regards,
Paul

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

paulvha, 2021-08-22:
in SPS30.cpp, line 482 I do a check whether the SPS30 was set to sleep earlier before performing a wake-up call. If you load a new program this is reset. For now comment out line 482        ` if (! _sleep) return(ERR_OK);`  to `//        if (! _sleep) return(ERR_OK);` this will bypass the check and sent the wakeup instruction.
regards,
Paul

ra1nb0w, 2021-08-23:
thank you @paulvha

closed 2022-05-24 by paulvha