paulvha/sps30 #18: CRC error when reading version info

uutzinger, 2021-01-01:
This is a suggested change to your code.

I attached sensor to ESP8266 i2c using D1,D2 but also tested with D3/D4 and D5/D6.
Its power is 5V and pullup is to 3.3V.

The sensor works with exception of reporting faulty version information.

I modified your code in uint8_t SPS30::I2C_ReadToBuffer(uint8_t count, bool chk_zero)
so that data is copied to the receive buffer before the CRC is calculated.
I recommend this modification so one can see what the transmitted version number is.

My sensor reports the following with your demo program  "Basic Readings with any i2c" when you implement the above change.

I2C Sending:  0xD1 0x00
I2C CRC error: Expected 0x69, calculated 0xF4
I2C Received: 0x02 0xFF length: 2

Error during reading from I2C: 0x51
Read version info with error.
Firmware level: 2.255
Library level : 1.4

Second byte of version information is likely faulty.

paulvha, 2021-01-02:
Thanks Ulf, 

Either the received data is good and can be used OR the data is not good and should be disgarded. 

For debug reasons I can see why one would implement a change like you propose and I have used that many times (e.g. see line 1594 in sps30.cpp). 

But in normal situation I would not try to inteprete data that has been received in error and make assumptions without having a deep understanding of the protocol.

I am not planning to change, however this is the charm of open-source so you tailor to your needs.:-)
regards,
Paul

uutzinger, 2021-01-04:
Understood.

In your extras folder you have a copy of an old data sheet.
The old data sheet does not mention an ability to read version information.
I might have an "old" sensor.

For now I will assume that a failure to read the version information implies that the sensor has functionality at firmware level V1.0. (= no sleep & wake).

paulvha, 2021-01-06:
HI Ulf,
The old datasheet was not complete. I got additional functions from studying some of Sensirion published code. If you run the example1: basic readings (or any of the other), it will display the firmware level. E.g. I have SPS30 with 1.0 version and 2.2 version and both will show the firmware level. If you have V1.0 indeed a number of functions will not be available.
regards,
Paul

closed 2023-01-26 by paulvha