paulvha/sps30 #20: Replacement Connector Found!

ccdzapper, 2021-06-19:
This is not an issue - just a note to help others that may have damaged or lost the flimsy cable with the ZHR-5 connector. It is available inexpensively on eBay: https://www.ebay.com/itm/114551266422
Make sure you pick the correct type, which is 1.25mm 5 Pin.
(10 Sets JST SH 1.0 ZH 1.5 PH 2.0 XH 2.5 Housing Connector Female  Male Wire)

paulvha, 2021-06-20:
thanks. good to know will add that in Readme

closed 2021-06-20 by paulvha

Nkae, 2021-11-10:
Hi!
I also found another compatible connector. I copy the link for Mouser as it is where I bought them. I also recommend buying spare crimp contacts as they are very small and fragile.

https://www.mouser.es/ProductDetail/Wurth-Elektronik/648005113322?qs=%2Fha2pyFaduguH2zIpdkgUWxmzUvrTES979PXEupx7lQusLC5mK%2FQfQ%3D%3D

https://www.mouser.es/ProductDetail/Wurth-Elektronik/64800113722DEC?qs=%2Fha2pyFaduguH2zIpdkgUUfd6dp6pTTujW8FuBzdSDO2pxvJN95p5w%3D%3D

paulvha, 2021-11-10:
thanks for the tip. Can you solder these or do you need a special tool to construct it together ?

Nkae, 2021-11-11:
Need a crimp tool. I use a Connector Pliers model PA-09 by Engineer I had from other works. But I'm not sure if the wires I'm using are a bit to big for the connector or the tool crashes too much the connector because after I crimp it I have to squeeze it vertically to enter in the plastic part.

It may be useful if special lengths are needed of there is no stock on JST ones (as now).

paulvha, 2021-11-11:
thanks I will add this in the readme !

regards,
Paul