paulvha/sps30 #21: PM1.0 measure over i2c sometimes give 0

hpsaturn, 2021-07-04:
Hi,

First, thanks for your library, for your work on it. I'm using this library on [CanAirIO firmware](https://github.com/kike-canaries/canairio_firmware#readme) and our wrapper library [CanAirIO Sensorlib](https://github.com/kike-canaries/canairio_sensorlib#readme). With UART implementation we don't have this issue but with i2c sometimes the PM1.0 value give 0, the other variables, PM2.5 and PM10 works fine.

When the debug flag is enable, before the issue is reached, we have the next output:

```python
20:37:03.820 > Error: Received NO bytes
20:37:03.820 > I2C Received: length: 0
20:37:03.820 > 
20:37:03.820 > Error during reading from I2C: 0x51
20:37:03.820 > -->[SPS30] read > done! 
20:37:03.820 > -->[SENSORS] PM1:000 PM25:008 PM10:008 CO2:0000 CO2humi:0.0% CO2temp:0.0°C H:39.21% T:25.5°C
```

It is weird because the error is detected by SPS30 library but something reseting to cero the PM1.0 value. The `read` method implementartion in our library is [this](https://github.com/kike-canaries/canairio_sensorlib/blob/0c53c8973cad22ebd41c9cb2ef454e87f7d53b9f/src/Sensors.cpp#L329-L361)

The final output after this issue is something like this:

[image]

Thanks

paulvha, 2021-07-05:
thanks for reaching out. I see 2 issues:
**1. Why doesn't it read correctly on I2C all the time ?** 
- It could be a wiring issue: is the sense wire connected to GND firmly are the other wires connected strong enough.
- missing / incorrect  pull-up resistors on SDA/SCL, 
- speed on I2C too high (max 100K is allowed)

**2. Error handling.** 
looking at the code in GetValues(), around line 932, there is a call : ret = I2C_SetPointer_Read(_I2C_Max_bytes); However, I do not check this return code, as I do with UART. I will upload a new version 1.4.11 in the coming minute.

regards,
Paul

hpsaturn, 2021-07-05:
Thanks for the fast respond to the question. I think that was the error, the handling error issue. After two hours the CanAirIO Bike version works fine. 

The only issue now is that the last version 1.4.11 wasn't in the Platformio registry, you maybe know if is missing the publication there? I tested it with the Github repo like a library target.

[image]

paulvha, 2021-07-05:
hi Antonio,

Don't know the Platformio registry, or how it gets there, but good to know it works for you.

Regards from the Netherlands.

Paul

hpsaturn, 2021-07-05:
Maybe your library was indexed before the last method of publish on PlatformIO and maybe it is on automatic mode, and I need only waiting for this. Then, no problem. Thanks

Regards from Berlin.

Antonio.

closed 2021-07-05 by hpsaturn