paulvha/sps30 #25: Compile error when excluding UART

davidjohngraham, 2022-01-04:
Hi Paul

When I comment out `#define INCLUDE_UART 1` I get a compile error:
_[...] /sps30.cpp:772:1: error: control reaches end of non-void function [-Werror=return-type]
  772 | }_

I'm compiling for a Particle Boron using Particle Workbench (MS Visual Code).

Thanks 

Dave

davidjohngraham, 2022-01-05:
I have added `return(0xff);` after line 763 in sps30.cpp, which fixes the compiler error. I'm not sure if this is the most appropriate error code to use?

paulvha, 2022-01-05:
HI Dave
I have just tried this on Arduino IDE and commenting out #define INCLUDE_UART 1 does not cause an error. I looked that the code. Actually, this must be a compiler error. The I2C code is returning an error if not successful  (unless you disabled I2C also ..) and the compiler should notice that. I know of others that use the SPS30 code , but never heard about this problem. What you could try instead of satisfying the compiler with a fake return, is to comment out the last lines in the subroutine that are handling the  UART. Just to see what happens

davidjohngraham, 2022-01-05:
Hi Paul

Yes, commenting out the lines that handle the UART also works.

I think the complier error may result from a change in version 2.2.0 of the Particle device os (see https://github.com/particle-iot/device-os/pull/2323).

My C++ skills are not sufficient to determine whether this is a genuine issue with your code, or in the Particle implementation.

davidjohngraham, 2022-01-05:
I'm guessing that the error results from the fact that no value is returned under the theoretically possible condition that UART is disabled and `if (_Sensor_Comms == I2C_COMMS)` is FALSE at line 735. I accept that this condition can't arise in practise if we're using i2c comms, but the compiler doesn't know that.

Happy to be corrected if I've misunderstood.

paulvha, 2022-01-05:
There is a check on BOTH UART and I2C begin disabled during compile to cause an error message, but indeed only in that situation it could happen. The compiler s pre-processor is in error in it should handle the #if defined rules correctly and include and exclude the right code as a result.  But there is always a work-around :-)

regards
Paul

closed 2023-01-26 by paulvha