paulvha/sps30 #26: Error reading sps30.

roberbike, 2022-01-27:
Hi.
Using the library with canairio proyect, I found that some times the sensor show me some errors:

->[SLIB] attempt enable sensor : SPS30
-->[SLIB] SPS30 Serial number :  3D59CD669A90A7EB
-->[SLIB] SPS30 product name :  00080000
-->[SLIB] SPS30 firmware level :  2.1
-->[SLIB] SPS30 Hardware level : 0
-->[SLIB] SPS30 SHDLC protocol : 0.0
-->[SLIB] SPS30 Library level : 1.4
-->[SLIB] SPS30 Detected SPS30 via I2C. 
-->[SLIB] SPS30 Measurement OK 
-->[SLIB] sensor registered : SPS30   :D

Here the error:

[E][SLIB] SPS30 error msg : Protocol error
-->[SLIB] AM2320 read : done! 
-->[HEAP] LOOP bytes used : 5860b/052Kb
-->[SLIB] Sensors devices count : 2 (SPS30,AM232X,)
-->[SLIB] Sensors units count : 6 (PM1,PM2.5,PM4,PM10,T,H,)
-->[SLIB] Preview sensors values : PM1:18.0 PM2.5:20.0 PM4:20.0 PM10:20.0 T:20.0 H:38.3 
-->[SLIB] SPS30 read  : done! 
-->[SLIB] AM2320 read : done! 
-->[HEAP] LOOP bytes used : -1768b/054Kb
-->[SLIB] Sensors devices count : 2 (SPS30,AM232X,)
-->[SLIB] Sensors units count : 6 (PM1,PM2.5,PM4,PM10,T,H,)
-->[SLIB] Preview sensors values : PM1:19.0 PM2.5:20.0 PM4:20.0 PM10:20.0 T:20.0 H:37.1 
-->[MQTT] Anaire cloud published : payload size: 197
-->[IFDB] CanAirIO cloud write : payload size: 48

other example with more info:
11:46:13.149 > -->[SLIB] trying to load I2C sensors..
11:46:13.155 > -->[SLIB] I2C SPS30 starting sensor..
11:46:13.155 > I2C Sending:  0xD1 0x00
11:46:13.160 > I2C Received: 0x02 0x01 length: 2
11:46:13.160 >
11:46:13.160 > I2C Sending:  0xD0 0x33
11:46:13.166 > I2C Received: 0x33 0x44 0x35 0x39 0x43 0x44 0x36 0x36 0x39 0x41 0x39 0x30 0x41 0x37 0x45 0x42 0x00 0x00 length: 18
11:46:13.171 >
11:46:13.171 > -->[SLIB] SPS30 Serial number :  3D59CD669A90A7EB
11:46:13.276 > I2C Sending:  0xD0 0x02
11:46:13.276 > I2C Received: 0x30 0x30 0x30 0x38 0x30 0x30 0x30 0x30 length: 8
11:46:13.276 >
11:46:13.276 > -->[SLIB] SPS30 product name  :  00080000
11:46:13.276 > I2C Sending:  0xD1 0x00
11:46:13.276 > I2C Received: 0x02 0x01 length: 2
11:46:13.276 >
11:46:13.276 > -->[SLIB] SPS30 firmware level:  2.1
11:46:13.276 > -->[SLIB] SPS30 Hardware level:  0
11:46:13.276 > -->[SLIB] SPS30 SHDLC protocol:  0.0
11:46:13.276 > -->[SLIB] SPS30 Library level :  1.4
11:46:13.276 > I2C Sending:  0xD3 0x04
11:46:15.196 > -->[SLIB] SPS30 Detected SPS30 via I2C. 
11:46:15.202 > I2C Sending:  0x00 0x10 0x03 0x00 0xAC
11:46:16.197 > -->[SLIB] SPS30 Measurement OK 
11:46:16.203 > -->[SLIB] I2C detected SPS30 sensor :)

And the error:

11:46:24.173 > -->[SLIB] AM2320 read > done! 
11:46:24.254 > I2C Sending:  0x02 0x02
11:46:24.259 > I2C Received: 0x00 0x01 length: 2
11:46:24.259 >
11:46:24.259 > I2C Sending:  0x03 0x00
11:46:24.261 > I2C CRC error: Expected 0xE8, calculated 0xD0
11:46:24.267 > I2C Received: 0x41 0x5B 0x50 0xFB 0x41 0x67 0xEB 0x2F length: 8
11:46:24.273 >
11:46:24.273 > Error during reading from I2C: 0x51
11:46:24.277 > [E][SLIB] SPS30 Protocol error
11:46:24.298 > -->[SLIB] PM1:000 PM25:000 PM10:000 CO2:0000 CO2humi:0.000000% CO2temp:0.000000°C H:37.099998% T:18.700001°C
11:46:24.309 > -->[SLIB] new sample time: 15
11:46:39.264 > I2C Sending:  0x02 0x02
11:46:39.265 > I2C Received: 0x00 0x01 length: 2
11:46:39.265 >
11:46:39.265 > I2C Sending:  0x03 0x00
11:46:39.267 > I2C CRC error: Expected 0xFF, calculated 0xF8
11:46:39.273 > I2C Received: 0x41 0x97 0xDB 0x34 0x41 0xA0 0x95 0x0E 0x41 0xA0 0x95 0x07 length: 12
11:46:39.284 >
11:46:39.284 > Error during reading from I2C: 0x51
11:46:39.284 > [E][SLIB] SPS30 Protocol error
11:46:39.290 > -->[SLIB] Any data from sensors? check your wirings!
11:46:39.314 > -->[SLIB] PM1:000 PM25:000 PM10:000 CO2:0000 CO2humi:0.000000% CO2temp:0.000000°C H:37.099998% T:18.700001°C
11:46:41.998 > -->[MQTT] Anaire sensor payload published. (size: 194)

I am not sure if this is an issue or not.
Thanks.

paulvha, 2022-01-28:
Hi,

Without additional information it is hard to give any advice.

What kind of board / hardware setup do you use ?
How do you connect?
How do you power ?
Does it work with a standard sketch?

regards,
Paul

roberbike, 2022-01-28:
> Hi,
> 
> Without additional information it is hard to give any advice.
> 
> What kind of board / hardware setup do you use ? How do you connect? How do you power ? Does it work with a standard sketch?
> 
> regards, Paul

Hi Paul.
The hasdware uswd is TTGO T-display (esp32) + sps30 by i2c+ AM2320 by i2c, with a battery connected to the board and the usb charguing and powering the hardware.
The power for sps30 is 5V, supplied from a step up, fron 3.3 V to 5 V.
The firmware used is https://github.com/kike-canaries/canairio_firmware that use the sensors library https://github.com/kike-canaries/canairio_sensorlib.

Your library works , but if debug is activated, the error messages displayed before, are displayed by serial debug.
When this happen , the values appearing are the same than the last value measured.

Thanks and regards.
Robert.

paulvha, 2022-01-28:
Hi Robert,

When a protocol error occurs either we did not get enough bytes or had CRC failures. Then the data that is provided should be neglected. 

I would suspect a hardware issue:
It could be the power not being stable enough. I got very weird results in the past when connecting to 3V3 instead of 5V. Not only needed for the fan, but also the internal regulator. Not sure how well the step up is working or how stable. Maybe adding a capacitor on the 5V would help. Also check that it is a clean power signal so it does not have spikes etc

The other aspect is pull-resistors on the I2C line. The AM2320 does not have those (according to the datasheet) and the SPS30 has very weak (~50K). Both SDA and SCL should have pull resistors to 3V3 ( each resistor 4K7 to 10k)

It could also be bad line connections. Seen that before with others that I have helped.

regards,
Paul

closed 2023-01-26 by paulvha