paulvha/sps30 #28: ERR_OK and ERR_TIMEOUT overwrite framework definitions

oseiler2, 2022-05-24:
I'm trying to use this lib with the Arduino framework on ESP32 in PlatformIO and get compile errors when also using the framework's lwip implementation (basically anything using sockets if I'm not mistaken) because sps30.h redefines ERR_OK and ERR_TIMEOUT that are already defined in lwip/err.h.
I'm not sure if there's an easy way to avoid this and appreciate any help.

[buildlog.txt](https://github.com/paulvha/sps30/files/8760639/buildlog.txt)
[Sample project showing the error](https://github.com/paulvha/sps30/files/8760641/SPS30.zip)
[esp-lwip](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/lwip/err.h)

paulvha, 2022-05-24:
it is a matter of definition. I have updated the SPS30 library and added "SPS30_" before each error code. Download the 1.4.14 version and it should work now.

regards,
Paul

oseiler2, 2022-05-24:
Awesome, thanks for the quick fix!

closed 2022-05-24 by oseiler2

oseiler2, 2022-05-25:
Quick follow-up: do you happen to know how to update the version for PlatformIO (still on 1.4.12), so that it can be pulled automatically?
https://registry.platformio.org/libraries/paulvha/sps30/versions

paulvha, 2022-05-25:
some time ago I was contacted by CanAirIO community with some questions. Apparently, they control the content on the link you provided. Try to contact them on [https://canair.io/](https://canair.io)
regards,
Paul

oseiler2, 2022-05-25:
Thanks, it's been updated now.