paulvha/sps30 #29: 'ERR_DATALENGTH'  'ERR_OK' not declared

MetalLocket0, 2022-06-29:
Hello Paul, I've been trying to get the SPS30 to read PM values using an Heltec ESP32 LoRa WiFi board, ive tried i2c and UART, yet am experiencing this issue, which is weird since it was not giving me the error of having things undeclared but rather saying that the sps30 probe was not reading. Any assistance would be helpful thank you.

MetalLocket0, 2022-06-30:
Hello, 

I was using the plotter example. Thank you again.

paulvha, 2022-07-01:
it looks to me that you have a combination of the  1.4.14 and earlier versions of the driver. In 1.4.14 I have changed the definition of the error message. E.g. from ERR_DATALENGTH to SPS30_ERR_DATALENGTH. Can you reinstall and check that?

MetalLocket0, 2022-07-05:
well i did all of that and it seems to not show that error anymore but now i have 'SP30_COMMS' was not declared as an error. I put the protocol as I2C and am using the same example.

MetalLocket0, 2022-07-18:
Thank you paul it now works! It is giving me "Due to I2C buffersize only the SPS30 MASS concentration is available !!!" 
anyway to get the other readings? 

Thank you,
Leal

paulvha, 2022-07-18:
Good to hear.. for the i2c buffer.. in de sps30.odt is a details description what you can do about it. The file can be read with nearly any wordprocessor..

Regards Paul

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

paulvha, 2023-01-26:
solved

closed 2023-01-26 by paulvha