paulvha/sps30 #30: Support for ESP32C3

hpsaturn, 2023-01-25:
## Overview

The library fails with **ESP32C3** over Espressif 5.0.0 and also over Espressif 6.0.0. With other variants, like ESP32, 8266 and also `Atmelsam` works fine. 

```cpp
Compiling .pio/build/ESP32C3/lib658/S8_UART/utils.cpp.o
.pio/libdeps/ESP32C3/sps30/src/sps30.cpp: In member function 'bool SPS30::setSerialSpeed()':
.pio/libdeps/ESP32C3/sps30/src/sps30.cpp:1097:13: error: 'Serial2' was not declared in this scope
             Serial2.begin(_Serial_baud);
             ^~~~~~~
.pio/libdeps/ESP32C3/sps30/src/sps30.cpp:1097:13: note: suggested alternative: 'Serial1'
             Serial2.begin(_Serial_baud);
             ^~~~~~~
             Serial1
```

### Dependencies:

```
Resolving ESP32C3 dependencies...
Platform espressif32 @ 6.0.0 (required: espressif32)
framework-arduinoespressif32 @ 3.20006.221224
sps30 @ 1.4.14
```

paulvha, 2023-01-26:
Thanks for the finding. I have corrected and just uploaded the new version 1.4.16. let me know

hpsaturn, 2023-02-14:
Hi again,

I did some extra tests with my old setup using only esp32 boards, and I found that this validation (`#if Serial2`) doesn't works for them. Validate with `Serial2` only works for ESP32C3, and for this reason we close this issue. But the right way to validate the Serial2 interface in the both architectures is using:

`#if SOC_UART_NUM > 2`  for the ESP32, because it resolve 3. For the ESP32-C3 resolve 2.

In my library I have a similar block:

```cpp
case SERIALPORT2:
        #if SOC_UART_NUM > 2
            DEBUG("-->[SLIB] UART COMM port \t: Serial2");
            if (pms_type == SENSORS::SSPS30)
                Serial2.begin(speed_baud);
            else
                Serial2.begin(speed_baud, SERIAL_8N1, pms_rx, pms_tx, false);
            _serial = &Serial2;
            break;
        #else
            DEBUG("-->[SLIB] Force UART port \t: Serial1");
            Serial1.begin(speed_baud, SERIAL_8N1, pms_rx, pms_tx);
            _serial = &Serial1;
        #endif
```

paulvha, 2023-02-14:
Don't understand the issue. I use `(#ifdef Serial2` ) not `(if Serial2`). If there is no Serial2 defined, I can not use it. 

In the sketch the right Serial port to use should be provided. As indicated in the Readme, I kept the "old" structure in for backward compatibility. For future implementation with serial, I would rather see that example12 setup for serial communication is used.
Much easier :-)

hpsaturn, 2023-02-14:
Sorry, I did a mistake, I want to say `ifdef`. In my previous version I was using your similar workaround (ifdef Serial2):

[image]

but it working with C3, but with ESP32 and Espressif 4.4.0 for example, this validation doesn't works.